ekknod/pcileech-wifi #1: Device Failure to start.

F0RSV1NNA, 2023-07-25:
Saw your post and def +rep. Went over the code a bit and decided to build it to test before i start modifying and hopefully pushing the full device emulation ball further for anticheat testing.

currently as is, with no modification when compiled the device has a code 10 failure to start like most broken FW on the market.

running the newest version of windows 11. which i suspect could be the issue right off the bat.

my system is as follows : 
13900k
z690 asus board
4090
LeetDMA

F0RSV1NNA, 2023-07-25:
An update:
After flashing your completely unmodified firmware (no changes made to your source.)
on an intel board, (z690 asus creator wifi)
Compiled in either Vivado versions 2021/2022/2023
or on windows 10 or windows 11.
My LEETDMA from Enigma either fails to fully "emulate" and all read pages fail.
or the OS will hang 2 seconds into boot.
OR it just doesn't post at all. Tried across all 3 separate commit versions of your SC.  

very unfortunate as its a great POC since it works for you.

b4b41, 2023-07-25:
I have the same with enigma-x1.. any additional steps we need to do?

cycript, 2023-07-25:
Your card failing to read any memory may be caused by the driver setting your device's power state to power off.

ekknod, 2023-07-25:
My system: i7 3770k. Z77 board. I will check tomorrow if there is anything i can do.

F0RSV1NNA, 2023-07-26:
> Your card failing to read any memory may be caused by the driver setting your device's power state to power off.

This was a really smart idea! i explored it making sure the power state was at max performance on all PCI devices aswell trying to make sure the driver wouldn't set the device to power off or anything like that and i couldn't find any evidence that it was the problem. 

i have however realized upon testing further that if i can get posted and into windows. if any program (aida64, RWE64bit, hwinfo, arbor, etc.) tries to scan the PCI devices on the system the system will instantly hard lock and take several hard restarts/resets before post (with the card unplugged completely) 
[image]

I managed to get this screen up in time before it crashed. Unclear if it could be useful to you ekknod and thanks for your work! truly impressive stuff. and a 3770k isn't your only system right?

ekknod, 2023-07-26:
[pcileech_squirrel_top.zip](https://github.com/ekknod/pcileech-wifi/files/12169677/pcileech_squirrel_top.zip)
That's my build, and it works. I suspect some of card calls are handled differently with different machines.

If you want to debug this issue -> Enable debug filter from registry:
[image]

Then run fun-hooker & reinstall driver:
[fun – hooker.zip](https://github.com/ekknod/pcileech-wifi/files/12169725/fun.hooker.zip)

streamable: https://streamable.com/a85k9s

ekknod, 2023-07-26:
that's build for squirrel, it's built from source without any changes. you could debug your issue with my provided tool.

ekknod, 2023-07-26:
https://github.com/ekknod/xilinx_cfg
[image]

F0RSV1NNA, 2023-07-26:
> [pcileech_squirrel_top.zip](https://github.com/ekknod/pcileech-wifi/files/12169677/pcileech_squirrel_top.zip) That's my build, and it works. I suspect some of card calls are handled differently with different machines.
> 
> If you want to debug this issue -> Enable debug filter from registry + reboot: [image]
> 
> Then run fun-hooker & reinstall driver: [fun – hooker.zip](https://github.com/ekknod/pcileech-wifi/files/12169725/fun.hooker.zip)
> 
> streamable: https://streamable.com/a85k9s

I suspect you are correct about it working differently on different machines. even attempting to debug with the firmware file you provided. my system hard freezes as soon as i attempt to "scan" the bus 2 device. 

i've tried lots of things and i suspect its just down to compatibility against my system.
Amazing work still though! This will definitely change the scene of DMA.

ekknod, 2023-07-26:
from my friend newer intel machine:
[image]

(0x7044) register returns: 0x00000000
(0x7044) register is handled by my firmware, and it should return 0x00000002 instead.
[image]

ekknod, 2023-07-26:
```
bar_read_write` <= (snoop_valid_bar_rd | snoop_valid_bar_wr) & (snoop_addr_id == `4'hF); 
```
 
This check validates if BAR is starting with "0xF  - - - "  e.g. my own BAR: 0xF780000
this is most likely the issue for everyone here, because this is not case with every system.

Replace this "entry" number with correct for your own system, and it should be good to go.

ekknod, 2023-07-26:
@F0RSV1NNA @cycript @b4b41

dom0ng, 2023-07-26:
Legendary.

cycript, 2023-07-26:
Legendary indeed. Thank you for your work, it is a great learning resource.

F0RSV1NNA, 2023-07-27:
> ```
> bar_read_write` <= (snoop_valid_bar_rd | snoop_valid_bar_wr) & (snoop_addr_id == `4'hF); 
> ```
> 
> This check validates if BAR is starting with "**0xF** - - - " e.g. my own BAR: **0xF**78F0000 this is most likely the issue for everyone here, because this is not case with every system.
> 
> Replace this "entry" number with correct for your own system, and it should be good to go.
> 
> This most likely fixes the issue for everyone. I will look if there is some better solution for this.

Continuing after alot of trial and error

My bar is 0x87C00004 
so i took the lastest version of your source and updated line 202 to reflect with
  My bar is 0x87C00004  so 0x8 = 8. = 4'h8 
  recompile. and the issue changed completely.
  
  now instead of getting the device loading but the driver failing i get stuck on a BSOD at windows startup 
  the reporting errors are: 
  "WHEA_UNCORRECTABLE_ERROR"
  "WHEA_INTERNAL_ERROR"
  
  i then figured it could've been an issue inside my z690 bios instead of an actual FW issue but more just being incompatible with my board/bios settings
  
  i attempted disabling every onboard device
  disabling TPM 2.0/Device security options 
  disabling com port
  disabling link state management
  
  Nothing i could do bios/setting wise was able to solve the bsod issues.
  i attempted flashing to a older bios revision. checking bar, trying again 
  flashing newer bios revisions. checking settings checking bar trying again.
  nothing i could do in the bios of the Asus Creator z690-pro-wifi could solve the BSOD issues.
  
  Im about to attempt again on a different z690 board and see if the issue repeats.
  will update with an edit with what i find.

xCodeZerox, 2023-07-27:
> > ```
> > bar_read_write` <= (snoop_valid_bar_rd | snoop_valid_bar_wr) & (snoop_addr_id == `4'hF); 
> > ```
> > 
> > 
> >     
> >       
> >     
> > 
> >       
> >     
> > 
> >     
> >   
> > This check validates if BAR is starting with "**0xF** - - - " e.g. my own BAR: **0xF**78F0000 this is most likely the issue for everyone here, because this is not case with every system.
> > Replace this "entry" number with correct for your own system, and it should be good to go.
> > This most likely fixes the issue for everyone. I will look if there is some better solution for this.
> 
> Continuing after alot of trial and error
> 
> My bar is 0x87C00004 so i took the lastest version of your source and updated line 202 to reflect with My bar is 0x87C00004 so 0x8 = 8. = 4'h8 compiled. and the issue changed completely.
> 
> now instead of getting the device loading but the driver failing i get stuck on a BSOD at windows startup the reporting errors are: "WHEA_UNCORRECTABLE_ERROR" "WHEA_INTERNAL_ERROR"
> 
> i then figured it could've been an issue inside my z690 bios instead of an actual FW issue but more just being incompatible with my board/bios settings since other people have mentioned success on boards that are not z690
> 
> i attempted disabling every onboard device disabling TPM 2.0/Device security options disabling com port disabling link state management
> 
> Nothing i could do bios/setting wise was able to solve the bsod issues. i attempted flashing to a older bios revision. checking bar, trying again flashing newer bios revisions. checking settings checking bar trying again. nothing i could do in the bios of the Asus Creator z690-pro-wifi could solve the BSOD issues.
> 
> Im about to attempt again on a different z690 board and see if the issue repeats. will update with an edit with what i find.

Same Issue herw on Aorus Z690 Elite AX. Maybe its related to the onboard Wifi?

F0RSV1NNA, 2023-07-27:
> Same Issue here on Aorus Z690 Elite AX. Maybe its related to the onboard Wifi?

i persoanlly dont see that being the issue,  but i could absolutely be wrong. i attempted disabling those devices inside the BIOS under "on board devices" and i had the same issues of BSOD on windows startup.  the only way could get into the OS was safe mode, if you picked safemode + networking the device loads and it bsod again. 

all of these systems have different ethernet controllers/wifi adapters if you look at the specifications on the boards or inspect them in device manager. so i cant say "its this device on your board"

also if your worried about having two wifi adapters that shouldn't be an issue either
i've ran more than 1 wifi adapter on my board in the past for hackntosh and i plugged it in earlier to check its BAR address and it ran perfectly fine along side my on board wifi controller

i tested just now on a different asus board z690-f strix and had it replicate the issue exactly on my main system. 
im guessing this is a specific z690 chipset  issue. i dont have any extra boards on hand to test if z790 is affected either.

ekknod, 2023-07-27:
My friend with i7 8700 got it working with these steps ->
- fix bar address
- unplug card
- reinstall windows
- install card back

F0RSV1NNA, 2023-07-27:
> * fix bar address
> * unplug card
> * reinstall windows

i went ahead and gave this an attempt just to see what would happen, as a result this is what i found:

Windows 10 
win 1809 - bsod
win 1909 - bsod
win 20h1 - bsod 
win 20h2 - bsod
win 22h2 - bsod 

went ahead and tried win 11 too for shits and giggles and i found it aswell experienced the "WHEA_UNCORRECTABLE_ERROR" & "WHEA_INTERNAL_ERROR" blue screen of death on windows startup (about 2 seconds into spinny circle)

im figuring the z690 chipset just throws a wrench into things when the bar is set correctly. i wish i could some how debug the device as its starting up to figure out exactly where its failing so i could provide more information.

Zoumax, 2023-07-27:
Got a "WHEA_UNCORRECTABLE_ERROR" on z790 ax + 13900k + win11 when booting after fix bar.

Maybe emulate not enough or wrong? I don't know what the emulate should behave like.

Zoumax, 2023-07-27:
I tried to use fun-hooker catching logs but failed. I'll try to find other ways to catch driver logs, then emulate another device. If someone can give me information or advice, I would appreciate.

xCodeZerox, 2023-07-27:
So we all have problems with 13gen or/and z690+ Chipset right?

F0RSV1NNA, 2023-07-27:
> So we all have problems with 13gen or/and z690+ Chipset right?

Yes.

ekknod, 2023-07-27:
> Continuing after alot of trial and error
>
> My bar is 0x87C00004 so i took the lastest version of your source and updated line 202 to reflect with My bar is 0x87C00004 so 0x8 = 8. = 4'h8 compiled. and [...]

since your base address register last 4 digits are not 0000, you have to change all offsets from my TLP handler.
[image]
that for example should be

```
if (snoop_addr_dw16[15:0] == 16'h0003) 
```

all of these addresses are multiplied by 4 (size of DWORD). e.g. h0003 = 3*4= 0x00C = (0x04 + 0x08)

ekknod, 2023-07-28:
Today i did modify code little, and it uses now relative addresses instead (should be more compatible). I would appreciate if you guys did test this new update (z690). 

@dom0ng @b4b41 @xCodeZerox @Zoumax @cycript @F0RSV1NNA

ekknod, 2023-07-28:
[prebuilt.zip](https://github.com/ekknod/pcileech-wifi/files/12193934/prebuilt.zip)

F0RSV1NNA, 2023-07-28:
> Today i did modify code little, and it uses now relative addresses instead (should be more compatible). I would appreciate if you guys did test this new update (z690).
> 
> @dom0ng @b4b41 @xCodeZerox @Zoumax @cycript @F0RSV1NNA

Sure ill test it right now!

F0RSV1NNA, 2023-07-28:
> prebuilt for people with squirrel [prebuilt.zip](https://github.com/ekknod/pcileech-wifi/files/12193934/prebuilt.zip)

So i tried this exact build you posted. aswell as a version i compiled from the source. 
they both reacted identically and these were the results : 

Flash card, Restart windows
Windows starts and gives the login
enter login password - CRASH instantly no blue screen just an instant crash.
computer restarts, loads into windows login screen,
login. Blue screen. same WHEA error. 
PC restarts, windows starts to startup and FREEZE on the spinny icon. no bsod just freeze.
manually restart PC and every boot attempt afterwards is the windows startup FREEZE with no bsod or error codes.

Then i tried something different. 
i unplugged the DMA card.
booted into windows.
and Hot plugged it into the system, with wind running.

This did not result in a crash but it also didn't "load" into windows. i was hoping this would allow me to grab the PCI config space with rwe and it did! 

this is what RW everything reported :
[image]

ekknod, 2023-07-28:
I assume z690 has then same issue as AMD. that's very unlucky.

dom0ng, 2023-07-28:
@F0RSV1NNA would you mind try on a fresh windows install ?

F0RSV1NNA, 2023-07-28:
> @F0RSV1NNA would you mind try on a fresh windows install ?
> 
> Also no device on device manager = can you try cold boot -> press power button for 10s to drain power from motherboard and device

i actually am trying on a fresh install now, and it presents the same issue.
(ended up putting a spare NVME drive into the system just so i can repeat flash windows instead of wrecking my main OS everytime) 

also i did attempt a cold boot after a hard windows freeze the 3rd boot. (psu unplug, hold button down, plug psu back in restart)

xCodeZerox, 2023-07-28:
> i actually am trying on a fresh install now, and it presents the same issue. (ended up putting a spare NVME drive into the system just so i can repeat flash windows instead of wrecking my main OS [...]

Same here

ekknod, 2023-07-28:
I would assume the issue is same as with AMD processors with newer gen intel.
no idea what causes it, just TLP not get delivered and PC freezes.

F0RSV1NNA, 2023-07-28:
> I would assume issue is the same as with AMD processors with newer gen intel (z690). no idea what causes it, just TLP not get delivered and PC freezes.

im not sure if this would help you or not, but here is the logs from event viewer.

[System logs.zip](https://github.com/ekknod/pcileech-wifi/files/12194527/System.logs.zip)

ekknod, 2023-07-28:
[pcileech_squirrel_top.zip](https://github.com/ekknod/pcileech-wifi/files/12194838/pcileech_squirrel_top.zip)
latest build with bug fix for some system picking up potentially default BAR instead of system BAR.

aa8u1, 2023-07-28:
> [pcileech_squirrel_top.zip](https://github.com/ekknod/pcileech-wifi/files/12194838/pcileech_squirrel_top.zip) 最新版本，修复了某些系统拾取潜在默认 BAR 而不是系统 BAR 的错误。

So, are AMD and 13th Gen processors supported?

F0RSV1NNA, 2023-07-28:
> > [pcileech_squirrel_top.zip](https://github.com/ekknod/pcileech-wifi/files/12194838/pcileech_squirrel_top.zip) 最新版本，修复了某些系统拾取潜在默认 BAR 而不是系统 BAR 的错误。
> 
> So, are AMD and 13th Gen processors supported?

Currently, not with that FW file, or the source compiled.

ekknod, 2023-07-28:
Updated issue title, no idea if older AMD's affected.

- ivy bridge (3770k, z77) works
- coffee lake (i7 8700) works
---------------------------------------------------------------------------------------
- AMD 5950x/5800x (x570,B550), not working
- Intel 13th Gen (z690), not working

F0RSV1NNA, 2023-07-28:
> Updated title
> 
> * ivy bridge (3770k, z77) works
> * coffee lake (i7 8700) works
> 
> * AMD 5950x/5800x (x570,B550), not working
> * AMD 1700x (B350), not working
> * Intel 13th Gen (z690), not working

12 gen and z790 aswell - not working

AAigars, 2023-08-02:
Seem to also be having the exact same issue as everyone else with i7-9700k (Z390)

KernelKrusha, 2023-08-02:
Same here with i9-10850k (Z490)

ekknod, 2023-08-02:
Updated the list. based on these reports ->
Intel Ivy Bridge - coffee lake (8th gen) only works.

Issue is in this TLP scoop system. for some reason in AMD processors only addresses (0x00,0x80,0x180) works.
I suspect same happens with Intel 9 - 13 gen.

https://github.com/ufrisk/pcileech-fpga/issues/133

Youssix, 2023-08-07:
Hey guys, can you share with me your project folder inside vivado after creating the project from the source pls.

ekknod, 2023-08-09:
@Youssix 
@AAigars
@dom0ng
@b4b41
@xCodeZerox
@Zoumax
@cycript
@KernelKrusha
@aa8u1
@F0RSV1NNA

woop woop. AMD support update is out. Hopefully it works for later gen intel as well.

dom0ng, 2023-08-09:
Wooop wooop

aa8u1, 2023-08-09:
goooooooooooooooooooooooooooooooooooooooooooood

AAigars, 2023-08-09:
Sadly, seems to still be freezing on boot for me on an i7-9700k.

ekknod, 2023-08-09:
I'm closing this issue, i assume everything works. Feel free to open new case in case its required to.