ekknod/pcileech-wifi #2: Am i missing the main config file ?

Youssix, 2023-08-07:
Hello,

Have a quick question, normally with my other firmware i do have a pcie_7x_0.xcl which allow me to configure a lot of things.

But with this firmware, after building i don't have this file.

[image]

Maybe it is because i'm using vivado 2023 ? 

i should install the 2022

F0RSV1NNA, 2023-08-08:
hey, i realize there isn't a ton of info out there about this, but this is a modified source from the original ulffrisk FPGA source.
you have full control over the config space with this source. edit everything with something like visual studio code (you can search for what you want to modify. as that config file no longer exists in this repo's source to put it simply.

ekknod, 2023-08-09:
Thanks F0RSV1NNA for your answer, accurate. I was busy with AMD support update.

closed 2023-08-09 by ekknod