ekknod/pcileech-wifi #3: Intermittent failures with MemProcFs

KernelKrusha, 2023-08-09:
Using MemProcFS, the latest version of the firmware on Intel i9-10850k (Z490) results in intermittent failures.

Log output:
`Device Info: FPGA: Bad PCIe TLP received! Should not happen!`

Other than this, it seems to be working perfectly.

AAigars, 2023-08-13:
Seem to be having an issue where the DMA card is unable to process any read/write operations for a few minutes at random intervals. Thought maybe it was something wrong with my setup or code but I flashed the original PCILeech firmware and I did not encounter this issue.

fortnitebott, 2023-08-13:
Same for me. Dma stop work for a few seconds after a certain number of seconds.  I think it's because of the network card driver. It has a Wi-Fi update interval setting.

ekknod, 2023-08-19:
> Same for me. Dma stop work for a few seconds after a certain number of seconds. I think it's because of the Qualcomm wireless card driver. It has a Wi-Fi update interval setting.

That's accurate. Multiplexer prefers signal from PCIe over FT601 currently.

jordandadams, 2023-08-26:
Could the DeviceID be why it's not passing Vanguard? Possibly a different device work?

ironman97, 2023-09-12:
So is there any better way to solve this problem?

ekknod, 2023-09-12:
is this actually problem?

ekknod, 2023-09-12:
is this actually problem?
it prefers device driver requests over FT601,

ekknod, 2023-09-12:
is this actually problem? it prefers device driver requests over FT601 on purpose.

AAigars, 2023-09-12:
> is this actually problem? it prefers device driver requests over FT601 on purpose.

Yeah, it made the firmware unusable for me. My read operation were taking way too long at certain times.

AboutElse, 2023-09-18:
About 
`Device Info: FPGA: Bad PCIe TLP received! Should not happen!`

I found that only the Intel (13900KS+Z790) platform will display these errors, and AMD (5950X+X570) seems to be work fine.

Alienccc, 2023-11-16:
> 大约 `Device Info: FPGA: Bad PCIe TLP received! Should not happen!`
> 
> 我发现只有Intel（13900KS+Z790）平台才会显示这些错误，而AMD（5950X+X570）似乎工作正常。

5900X+B550   This problem still occurs

uk0, 2023-11-20:
Z690 update bios  nice job.

ekknod, 2024-01-17:
Core is now updated Version. 4.13. Issue should be now fixed. Disable WIFI from system tray in case you have TLP misses.

closed 2024-01-17 by ekknod