ekknod/pcileech-wifi #4: Intermittent failures

Forever-APEX, 2023-08-09:
Device Info: FPGA: Bad PCIe TLP received! Should not happen! 
is this problem cased by 
[image] ?
I have a file for pcileech_squirrel. Test Vanguard with win10 1909-22h2 works well.
[for pcileech_squirrel.zip](https://github.com/ekknod/pcileech-wifi/files/12305105/for.pcileech_squirrel.zip) later I will test on win11 22h2
but how is available on win10?

KernelKrusha, 2023-08-09:
Why would you open a new issue when the **only** other open issue is identical?
https://github.com/ekknod/pcileech-wifi/issues/3

ekknod, 2023-08-09:
Vanguard patched it Yesterday in Europe. It should not be working.

plastik88, 2023-08-09:
Hey there ! very nice project. I have built the very latest from your repo. It shows up as a properly functioning Qualcom atheros wifi device with "this device is working properly". Good job!! However when starting vanguard AC with game I can no longer read. Should I be using an older build?

dom0ng, 2023-08-09:
@plastik88 Vanguard patched it 07/08/2023 in Europe around 16:00 (GMT + 1).

plastik88, 2023-08-09:
@dom0ng I am NA so it must not be just there

Forever-APEX, 2023-08-10:
> Vanguard patched it Yesterday in Europe. It should not be working.

Hello, thank you for your reply. I am even more curious about why I can use the bin I posted，I use dma cheat normally in Valorant with windows11 22H2

Forever-APEX, 2023-08-10:
> > > Vanguard patched it Yesterday in Europe. It should not be working.
> > 
> > 
> > Hello, thank you for your reply. I am even more curious about why I can use the bin I posted，I use dma cheat normally in Valorant with windows11 22H2
> 
> Hello
> 
> I did have taken a look at the firmware you posted.
> 
> It is not based on pcileech-wifi.
> 
> firmware posted is messing with some config space baseptr & nextptr. (I can be more precise if you wish) that basically fuck up the way most software parse the config space & device capabilities etc （ drvscan, vgk, SIV ex )...
> 
> That why it works. It's basically same method used by most of "working" firmware you can found on the market in the actual state.
> 
> AFAIK I think your question don't have answer because this is not related to pcileech-wifi :D

thanks

closed 2023-08-10 by Forever-APEX

Alienccc, 2023-11-17:
> > > Vanguard 昨天在欧洲修补了它。它不应该工作。
> > 
> > 
> > 您好，感谢您的回复。我更好奇为什么我可以使用我发布的垃圾箱，我在 Windows11 22H2 的 Valorant 中通常使用 dma 作弊
> 
> 你好
> 
> 我确实看了一下您发布的固件。
> 
> 它不是基于 pcileech-wifi。
> 
> 发布的固件扰乱了某些配置空间 baseptr 和 nextptr。 （如果你愿意，我可以更精确）这基本上搞砸了大多数软件解析配置空间和设备功能等的方式（drvscan、vgk、SIV ex）......
> 
> 这就是它工作的原因。这与您在实际状态下可以在市场上找到的大多数“工作”固件使用的方法基本相同。
> 
> AFAIK 我认为您的问题没有答案，因为这与 pcileech-wifi :D无关

How to get a reliable firmware