ekknod/pcileech-wifi #5: fun - hooker not working

get1n, 2023-08-26:
First of all nice release @ekknod :)

I tried to run your fun - hooker but unfortunately it doens't work; After reinstalling driver it only prints out 640_done:
[image]

settings are:
Capture kernel, Enable verbose, Pass-Through and Capture events.

What am I missing?

dom0ng, 2023-08-26:
@get1n 

You need to create this key to see additional informations.

HKEY_LOCAL_MACHINE\SYSTEM\CurrentControlSet\Control\Session Manager\Debug Print Filter The set the Default Value to 0xf. 

[image]

get1n, 2023-08-26:
I have :( @dom0ng 

[image]

ekknod, 2023-08-29:
> First of all nice release @ekknod :)
> 
> I tried to run your fun - hooker but unfortunately it doens't work; After reinstalling driver it only prints out 640_done: [image]
> 
> settings are: Capture kernel, Enable verbose, Pass-Through and Capture events.
> 
> What am I missing?

fun-hooker is installing PsSetLoadImageNotifyRoutine, that routine is looking for the target driver being loaded and then patches functions to provide debug information.
Could the driver name be different with your system for unknown reason?

ekknod, 2023-09-04:
I found source code in case someone needs https://pastebin.com/fNnyWKVj

closed 2024-08-21 by ekknod