ekknod/pcileech-wifi #6: May I ask how you obtained this data

1095150228, 2023-10-11:
May I ask how you obtained this data

`    else
    begin
      if (rd_addr[10:9] == 2'b01)
        begin
          case ({rd_addr32, 2'b00})
            {base_address_register + 16'h2000} : begin data_8 <= #TCQ 1; rd_data_raw_o <= #TCQ 32'hDEADBEEF; end
            {base_address_register + 16'h2200} : begin data_8 <= #TCQ 2; rd_data_raw_o <= #TCQ 32'hDEADBEEF; end
            {base_address_register + 16'h2204} : begin data_8 <= #TCQ 3; rd_data_raw_o <= #TCQ 32'hDEADBEEF; end
            {base_address_register + 16'h2208} : begin data_8 <= #TCQ 4; rd_data_raw_o <= #TCQ 32'hDEADBEEF; end
            {base_address_register + 16'h220C} : begin data_8 <= #TCQ 5; rd_data_raw_o <= #TCQ 32'hDEADBEEF; end
            {base_address_register + 16'h2210} : begin data_8 <= #TCQ 6; rd_data_raw_o <= #TCQ 32'hDEADBEEF; end
            {base_address_register + 16'h2218} : begin data_8 <= #TCQ 7; rd_data_raw_o <= #TCQ 32'hDEADBEEF; end
            {base_address_register + 16'h221C} : begin data_8 <= #TCQ 8; rd_data_raw_o <= #TCQ 32'hDEADBEEF; end
            {base_address_register + 16'h2220} : begin data_8 <= #TCQ 9; rd_data_raw_o <= #TCQ 32'hDEADBEEF; end
            {base_address_register + 16'h2224} : begin data_8 <= #TCQ 10; rd_data_raw_o <= #TCQ 32'hDEADBEEF; end
            {base_address_register + 16'h2228} : begin data_8 <= #TCQ 11; rd_data_raw_o <= #TCQ 32'hDEADBEEF; end
            {base_address_register + 16'h4020} : rd_data_raw_o <= #TCQ 32'h001800FF;
            {base_address_register + 16'h4028} : rd_data_raw_o <= #TCQ 32'h00000060;
            {base_address_register + 16'h4038} : rd_data_raw_o <= #TCQ 32'h00000002;
            {base_address_register + 16'h407C} :
              begin
                case (data_8)
                  1  : rd_data_raw_o <= #TCQ 32'h0000A55A; // EEPROM MAGIC
                  2  : rd_data_raw_o <= #TCQ 32'h00000004; // EEPROM_SIZE
                  3  : rd_data_raw_o <= #TCQ 32'h0000FFFB; // EEPROM_CHECKSUM
                  4  : rd_data_raw_o <= #TCQ 32'h0000E00E; // EEPROM version + revision
                  5  : rd_data_raw_o <= #TCQ 32'h0000E00E; // EEPROM_ANTENNA (2.4ghz, 5.0ghz)
                  6  : rd_data_raw_o <= #TCQ 32'h00000000; // EEPROM_REGDOMAIN (location data)
                  7  : rd_data_raw_o <= #TCQ 32'h00009C64; // EEPROM_MAC0 (64:9C)
                  8  : rd_data_raw_o <= #TCQ 32'h00000881; // EEPROM_MAC1 (81:08)
                  9  : rd_data_raw_o <= #TCQ 32'h0000C0C4; // EEPROM_MAC2 (C4:C0)
                  10 : rd_data_raw_o <= #TCQ 32'h00000100; // EEPROM_RXTX (00,01)
                  11 : begin rd_data_raw_o <= #TCQ 32'h00000000; data_8 <= #TCQ 0; end
                  default : rd_data_raw_o <= 32'h00000000;
                endcase
              end`

ekknod, 2023-10-12:
I did reverse engineer original card driver, and built correct replies.
Linux open-source driver was also good help.

1095150228, 2023-10-12:
这是来自QQ邮箱的假期自动回复邮件。
 
您好，我最近正在休假中，无法亲自回复您的邮件。我将在假期结束后，尽快给您回复。

closed 2023-10-12 by 1095150228