ekknod/pcileech-wifi #9: Issue with FPGA PCIe TLP Error Handling

uk0, 2023-11-15:
Hello ekknod,

Firstly, I would like to extend my gratitude for your efforts in maintaining the pcileech-wifi project. It's been an invaluable resource for the community.

I'm reaching out to report a recurring issue that I've encountered, which I believe might be of significance to other users as well. The issue pertains to an error message that is displayed under certain conditions: "Device Info: FPGA: Bad PCIe TLP received! Should not happen!"

Here's a brief description of the context in which the error occurs:

- 
Thank you for your time and consideration. I'm looking forward to your thoughts and any potential solutions you might suggest.

Best regards,
[github.com/uk0]

Alienccc, 2023-11-16:
我也是这个问题

uk0, 2023-11-17:
> 我也是这个问题

我再看看怎么修复 应该可以修复

Alienccc, 2023-11-17:
> > 我也是这个问题
> 
> 我再看看怎么修复 应该可以修复
有别的联系方式吗

uk0, 2023-11-21:
升级了bios 解决了

closed 2023-11-21 by uk0