ekknod/pcileech-wifi #11: Device Info: FPGA: Bad PCIe TLP received! Should not happen!

Supergun777, 2023-12-30:
Hello. I compiled and flashed the board without changing the code, and it gives me an error within a minute of work:
Device Info: FPGA: Bad PCIe TLP received! Should not happen!

Windows 11 Pro 22H2 22621.2861
11th Gen Intel(R) Core(TM) i5-11600K
Gigabyte Z590 GAMING X(rev. 1.0) Bios latest

Supergun777, 2023-12-30:
[HOMEZZZ.LOG](https://github.com/ekknod/pcileech-wifi/files/13798740/HOMEZZZ.LOG)

Supergun777, 2023-12-30:
[Uploading HOMEZZZ1.LOG…]()

Supergun777, 2023-12-30:
[HOMEZZZ1.LOG](https://github.com/ekknod/pcileech-wifi/files/13798768/HOMEZZZ1.LOG)

Supergun777, 2023-12-30:
[HOMEZZZ.LOG](https://github.com/ekknod/pcileech-wifi/files/13798777/HOMEZZZ.LOG)

Supergun777, 2023-12-31:
I solved the problem, you just need to turn off Wi-Fi in the tray next to the clock.

ekknod, 2024-01-17:
Core is now updated Version. 4.13. Issue should be now fixed. Disable WIFI from system tray in case you have TLP misses.

closed 2024-01-17 by ekknod