ekknod/pcileech-wifi #12: Encountering an issue

KendallCuen, 2024-01-18:
I apologize for asking such a foolish question, but I would like to know why, when I use your latest version for building (without making any modifications), the build result that I flash into my luker doesn't have any response after I start my computer. The network device that everyone sees is not listed in my Device Manager, and I'm unable to use it.

dexus1337, 2024-01-18:
Project is not fully updated/merged. The cfgspace is not modified, therefore the system cannot properly identify the device atm. You can fix it by re-adding the pcie_7x_0_* files from previous builds back to the project.

KendallCuen, 2024-01-18:
> 项目未完全更新/合并。由于cfgspace未修改，系统无法正确识别atm设备。您可以通过将先前版本中的 pcie_7x_0_* 文件重新添加回项目来修复该问题。

Thank you, I will try again.

KendallCuen, 2024-01-18:
> Project is not fully updated/merged. The cfgspace is not modified, therefore the system cannot properly identify the device atm. You can fix it by re-adding the pcie_7x_0_* files from previous builds back to the project.

Thank you, I will try again.

ekknod, 2024-01-18:
Project is fully updated, and pcie_7x_0.xci should set correct VID/PID automatically.

dexus1337, 2024-01-18:
> Project is fully updated, and pcie_7x_0.xci should set correct VID/PID automatically.

does it set capabilities etc too? Cause i think that might be his issue

ekknod, 2024-01-18:
What do you mean by that?

ekknod, 2024-01-18:
What do you mean by that?
.xci configures BAR and device details as well.

closed 2024-01-18 by ekknod

KendallCuen, 2024-01-18:
> What do you mean by that? .xci configures BAR and device details as well.

I believe that the lurker is not compatible with these firmware files, and I'm also unsure of the firmware it is actually compatible with because the file provided by the seller is named "pcileech_pciescreamer_top.bit."

I attempted to build using the pciescreamer version, but I found that it doesn't work properly after flashing it.

Normally, when the lurker is functioning correctly, there is a blinking blue LED.  However, after flashing it with the firmware you provided, it doesn't work as expected.  I'm not sure what the reason for this is.

Nonetheless, I appreciate your efforts, and I think I should consider switching from the lurker to leetdma or another device.