ekknod/pcileech-wifi #15: Can't access the system after flash

qinghemuyu, 2024-02-16:
Stuck on the startup screen, Act II.

qinghemuyu, 2024-02-16:
My devices are i7 8700 and ASUS B360M

F0RSV1NNA, 2024-02-17:
Start by testing, Unplug card.
does system boot? 

plug card in. power on
can you go to bios?
In bios? flash card with known working firmware.

system boots?  great.

its hard to say why your system isn't booting from my limited knowledge but id argue its because you have a B360M.

qinghemuyu, 2024-02-17:
I can go into the bios and have it enter the system properly when I reflash the other firmware. Thank you I'll try everything you said first.

qinghemuyu, 2024-02-17:
> 从测试开始，拔掉卡。 系统是否启动？
> 
> 插上卡，开机 可以进bios吗？ 在BIOS里？具有已知工作固件的闪存卡。
> 
> 系统启动？伟大的。
> 
> 根据我有限的知识，很难说为什么你的系统无法启动，但我认为这是因为你有 B360M。

I don't have working hardware on my device that compares to this firmware!

I can go into the bios and have it enter the system properly when I reflash the other firmware. Thank you I'll try everything you said first.

qinghemuyu, 2024-02-17:
> 从测试开始，拔掉卡。 系统是否启动？
> 
> 插上卡，开机 可以进bios吗？ 在BIOS里？具有已知工作固件的闪存卡。
> 
> 系统启动？伟大的。
> 
> 根据我有限的知识，很难说为什么你的系统无法启动，但我认为这是因为你有 B360M。

I can go into the bios and have it enter the system properly when I reflash the other firmware. Thank you I'll try everything you said first.

I don't have working hardware on my device that compares to this firmware!

我可以进入 BIOS，让它在重新刷新其他固件时正常进入系统。谢谢你，我会先试试你说的所有方法。

我的设备上没有能与该固件相比的硬件！

closed 2024-02-19 by ekknod