ekknod/pcileech-wifi #22: How to get register addresses?

Jotalz, 2024-04-22:
Hello,ekknod,If I use a different network card device,Which values need to be modified  in this [part](https://github.com/ekknod/pcileech-wifi/blob/690eafc88a784a55b0af042dedbddcc495f778fc/src/pcileech_tlps128_bar_controller.sv#L848) by myself?
I read some simple tutorials beforehand,they said I need obtaining the register addresses for the device by wiki or ida,but no specific acquisition method was pointed out. For example, if I use ida pro to reverse drive,What clue do I use to locate this value?

SmileHalo, 2024-04-29:
read this module [https://github.com/ekknod/pcileech-wifi/blob/690eafc88a784a55b0af042dedbddcc495f778fc/src/pcileech_tlps128_bar_controller.sv#L803](url)
and ath driver source to simulate tlp packet i guess.. it's hard to use ida reverse .sys binary file.  
you can find some public open-source or reverse-engineered driver here  [https://en.wikipedia.org/wiki/Comparison_of_open-source_wireless_drivers](url)

Jotalz, 2024-04-30:
To be honest, I can't understand these lists given by wiki,It lists a lot of drivers but only lists them I think,It lists them to indicate that they exist as open source,do I need other ways to find the driver source code?

SmileHalo, 2024-04-30:
ath.sys [https://elixir.bootlin.com/linux/latest/source/drivers/net/wireless/ath/ath9k](url)

Jotalz, 2024-04-30:
> ath.sys [https://elixir.bootlin.com/linux/latest/source/drivers/net/wireless/ath/ath9k](url) its linux source but for same device tlp packet is same. you can reference this source and fpga tlp bar controller module part to understand how the driver init on windows.

Very useful, thank you for your patience in giving me answers!

closed 2024-04-30 by Jotalz